BCN3D/Cura #148: Small Z-gap makes Cura wild (and insane)!

S3D-r, 2020-01-22:
**Application Version**
3.0.0.  (and earlier, I believe)

**Platform**
Win10-64bit

**Steps to Reproduce**
Two separate bodies (yet one STL), with small (0.4 mm) clearance between them in Z direction.  

**Actual Results**
The build is like a maniac: Top is built lower than supposed, then a couple of _infill_layes are built on top of that, then support material on top of that infill, then some more infill on top of that support, then bottom layers on top of that and then infill (as normal).  If the Z-gap is big enough (tested with 1.0 mm), the Cura performs OK.

You get the best idea by looking at the videos:
Wild and insane (0.4mm gap) : https://youtu.be/WwLwpGzXn54
OK (same item but 1.0 mm gap): https://youtu.be/8HLm2VZWoC0

eloipr, 2020-02-12:
Thanks for your feedback! We are trying to reproduce the bug to find a fix.

S3D-r, 2020-12-04:
So, did you try? Did you succeed?