BCN3D/Cura #149: Faulty material profile / fan blows at 1st layer

S3D-r, 2020-02-26:
**Application Version**

Latest.

With PLA the print cooling fan should activate at 2nd layer and above only, not at 1st layer. And that's how it normally works. But, when you print duplication mode and if you have non-level left/right nozzle (no shims used), the upper nozzle prints extra layers. So far so good. The lower nozzle will join the game later, and that's when the error happens:  The print cooling fan starts blowing immediately, cooling the very first layer. Should wait until the second layer.

mteterina, 2020-02-27:
Can you give me more information on what material and what nozzle you are using?

S3D-r, 2020-02-27:
PLA.  Left/Right  e3D  0.4mm.

S3D-r, 2020-02-28:
Hi!  I haven't modified these. I made further tests with a fresh installation of ver. 3.2.0,  and it seems like the first (upper) extruder starts to run the fan in the middle of the distortion layer. And the second starts to run the fan already when it's still above the purge bin (before the first layer), i.e. the extruder hasn't even moved over the print bed.

S3D-r, 2020-02-28:
Please make a similar test, it's easy to do.

mteterina, 2020-02-28:
I will take a look at it!

S3D-r, 2020-03-15:
Take your time (seems like you will do...).

mteterina, 2020-03-16:
Dear gentlemen, i don't know if you are aware of the situation that is occurring for the moment around the world. Due to the COVID-19 virus the whole country of Spain (where BCN3D is located) is in quarantine. This really effects our workflow and we have to reprioritize the type of work that we do. I have this issue noted on my to do list so i wont forget about it. I hope that you can be more lenient towards why this issue may take some time. Thank you for your understanding!

S3D-r, 2020-03-16:
Please accept my apologies for lack of courtesy and for weak situation awareness there in Spain. This Cura issue definitely isn't in top priority, not here in Finland nor there in Spain.  All my empathy to people around world who suffer from this nasty disease.