BCN3D/Cura #150: Color indication for extruder misleading when option "not overrridden" is chosen.

S3D-r, 2020-02-26:
**Application Version**
3.1.0 and all older versions.

You have options "Left Extruder", "Right Extruder" and "Not Overridden" for many parameters, like _Infill Extruder_, _Built Plate Adhesion Extrude_r, _Support Extruder_ etc...  If you choose Left Or Right, the will be an indication dot of the same color than the corresponding extruder material color is.  But, if you choose "Not Overridden", the color of the is always white. So, you can't conclude from the dot color, which extruder fill be used for that line type. And more complicated: If either of your filament colors happens to be white, you might jump to conclusion that this extruder will be used (cause the indication dot is white).

It would be better, if the indication dot would be same color than the extruder that will be used.
[image]

diegopradogesto, 2020-02-27:
In this case, the "Not override" option is always white, despite the color of the materials (not saying that it's not confusing when you have a white material in one of the extruders). "Not override" means that the same extruder used to print the model will be used for that specific part of the print (let's say walls in your example).

[image]

S3D-r, 2020-02-28:
How do you define, which of the extruders you can call "the one that builds the model"...?  Is it the Outer Wall Extruder, Inner Wall Extruder, Infill Extruder, Top/Bottom Extruder etc... (there are more options and each of these can be either Left or Right extruder). All these are used in building the model.  It's true that in most cases all of these are the very same extruder - but not always. 

But anyway, the clearest solution was if the "Not Overridden" spot was the same color as the extruders that it refers to. 
The second best would be some symbol that doesn't refer any color (like now it refers to color "white").
The worst is the current solution (white spot).

diegopradogesto, 2020-02-28:
The way you choose "which extruder to use to print which model" is by clicking in the two options that you see in the picture below:
[image]

Or right clicking on the model and select the extruder:
[image]

This will be the option by default ("Not override"). But afterwards you could select one extruder or the other to print specific parts (such as the walls, infill, support interface, and so). That's why you have the Extruder selectors in some of the settings.

In the example below I have selected the default extruder to be the right (green), but then I've selected the left one (black) to print the infill:
[image]

I hope it helps.

S3D-r, 2020-02-28:
Of course...  thanks for clarifying this (which now seems pretty obvious...).