BCN3D/Cura #151: Cura allows max 0.3mm layer height

S3D-r, 2020-02-26:
3.1.0 and older

For some reason, Cura seems to allow max 0.3mm layer height. Yet you may have a 1.0mm nozzle. A rule of thumb is, that a layer height should not exceed 80% of the nozzle size. I.e 1.0mm nozzle could have 0.8mm layer height. Yes indeed - but not with Cura. 

My opinion is, that Cura should only warn about high layer heights, not completely ban them.

closed 2020-02-27 by eloipr

S3D-r, 2020-02-27:
???   Closed without any comment?

eloipr, 2020-02-27:
You're right, the layer height shouldn't be banned. It has been fixed and will be available in the next release. It will only show a warning.

S3D-r, 2020-02-27:
Good to know, thanks. But why did you close the issue without commenting, and why did the issue not get any tag?

commit efc76ff1 (2020-02-27): fixes #151