BCN3D/Cura #152: Idle temperature applied not before the first cycle of "idle" nozzle.

S3D-r, 2020-03-05:
**Application Version**
3.2.0 and older

**Platform**
Win10-64 / e3D nozzle L=0.4, R=0.6

**Steps to Reproduce**
I have a dual-color print, where the Right Extruder ("R") participates only in the middle of the work (~3 hours from the beginning). So, the R nozzle won't be needed for hours.

**Actual Results**
After the bed is heated, the printer starts heating both extruders to their full temperature (not to the standby temperature).  

**Expected results**
The idle nozzle heater should first heat to idle temp and then wait until it's the right time to start heating to full temp (~1-2 minutes before swap). 

**Additional Information** 
It seems like Cura activates the idle temp after the first cycle of that particular nozzle. It should activate it right at the beginning of the print work.

The problem is not that bad if you print skirt/brim with both heads (as the first cycle will then be relatively soon), but if you don't, then the first cycle can be hours away, and the filament will burn. 

Forcing the "idle" nozzle to print skirt/brim might show as a good workaround for this bug, but it isn't, in many cases: The first layer will usually be printed at a slower speed. There's also more stuff to print at the first layer: brim/skirt. In the case of a large item, this all means that the first layer may take tens of minutes to print. So, even with this workaround, the "idle" nozzle will burn the filament in the hotend for a very long time.

I have raised this same issue more than 2 years ago ( https://github.com/BCN3D/Cura/issues/26 ), still not solved (yet the workaround was suggested at that time).

S3D-r, 2020-03-15:
No comments from BCN3D.

Well, outside of this forum I've discussed with SAT, and I now share their response so that a larger audience will benefit from it: 

**BCN3D:**  "_Many thanks for reporting this, I will transfer it to the software guys and see what they can do about it. Maybe in a next release this bug is fixed. For the moment, make sure to print with a skirt in dual, that way both hotends will start a normal cycle and the idle temperature will be set from the beginning of the print. If you don't want to wait so much, you can always set a higher speed for the skirt and first layer in Cura_."

diegopradogesto, 2020-04-01:
Well, I would say that if it's the first time the hotend is heated up, it shouldn't go to the standby temperature from the beginning. What CuraEngine (actually) does is to calculate how long it takes to heat up to the printing temperature and start heating some gcode lines before start using the other hotend, so the right temperature is set at the right time.

The weird thing here is that it heats up at the beginning. If that is the case, it's something to look at.

S3D-r, 2020-04-01:
Yes, that's the case.  BCN3D is investigating this (as can be read in my previous followup).