BCN3D/Cura #154: Print Sequence not adjustable if printing with right extruder

S3D-r, 2020-04-01:
-->

**Application Version**
3.2.0 and older

**Steps to Reproduce**
Put several items to platform, select them all, and pick left extruder as printing extruder. Go to "Special Modes" and you'll see "Print Sequence" as one option (assuming it's marked visible in settings visibility list). Now, go and select the right extruder as printing extruder. Voilà - the "Print Sequence" option disappears. It's magic!  (or a bug, would someone else say...).  

So, you can't print "One at a Time" with the right extruder.

diegopradogesto, 2020-04-06:
As far as I know, in Cura (even the original from Ultimaker) the print sequence is only available when there is one extruder. So the bug appears in the first place where you are able to select the setting with the left extruder. I checked and I couldn't reproduce it: "Print sequence" is not an active setting neither with the left nor the right extruder.

S3D-r, 2020-04-06:
Are you saying, that Cura shouldn't let **any** dual-extruder printer to print "One at a Time"? Why not, a "One at a Time" printing is in many cases very useful and from a technical point of view there's no difference if the printer is a single-extruder or dual-extruder.

diegopradogesto, 2020-04-06:
Well, for a dual extrusion one-at-a-time, the collision algorithm is much more complex, because you need to know what was already printed with one extruder in order to move the other to the start position without colliding. I say that is not impossible but afaik is not yet implemented in either Cura. In the Ultimaker's Cura, you can disable one extruder and then use the print sequence. But the "disable extruder" feature was introduced in later versions than the 3.2 in which BCN3D's Cura is based on.

S3D-r, 2020-04-06:
Not sure if you wrote a typo or not, but I wasn't talking about _dual extruding_ but _single extruding_ with a __dual-extruder printer_.

S3D-r, 2020-04-06:
And more. I don't see it much more complicated to write an algorithm for dual extruding than single extruding. The math is exactly the same for both extruders, the second extruder is only some centimeters further in x-direction than the first one - that's all.  Anyway, a lot of responsibility is left to the user for determining whether the collision will happen or not, and Cura assists the user by shadowing the needed space in the platform. I've used BCNCura a lot for One-at-a-time printing with Sigma, obviously with left head, and never had any problems with collisions.

diegopradogesto, 2020-04-06:
I was trying to say in _dual-extruder printer_, even if it's single extruding or dual extruding. Setting visibility does not react differently whether you're printing with only one or two extruders in the same job or not. Actually it behaves always as if you are printing with the two extruders even if you don't use both.

That's why if you select (for instance) in Cura PLA (extruder 1) and ABS (extruder 2), together, the build plate temperature takes the maximum of both values (there is a formula indicating so: https://github.com/BCN3D/Cura/blob/efc76ff1548ecb9a4dc8798fdc9e215e6f8d9780/resources/definitions/fdmprinter.def.json#L1967). Although you are only printing a model with PLA, Cura uses the formula and sets the temperature to the max value.

Same with the Print Sequence. It's a setting that should only be visible when there is only one extruder (https://github.com/BCN3D/Cura/blob/efc76ff1548ecb9a4dc8798fdc9e215e6f8d9780/resources/definitions/fdmprinter.def.json#L5445). That's why I say two things:
- There is a bug on that setting.
- The feature "enable/disable extruder" is very useful (introduced in Ultimaker Cura in 3.3.0: https://github.com/Ultimaker/Cura/releases/tag/3.3.0)

S3D-r, 2020-04-13:
I still disagree with your claim that print sequence should be visible with single extruder printers only. I've earlier explained, why it works very well with dual extruders, too.  The "enable/disable extruder" feature might be very useful in many cases, but for this issue, it's not needed.