BCN3D/Cura #156: No printer connected

civgmh, 2020-07-02:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

civgmh, 2020-07-02:
Application Version cura BCN3D 3.2.0
Platform Win 10
Expected results connect to network Epsilon printer
Actual Results: "no printer connected"

Additional Information:

see attached .
[image]

pedro-iatzky, 2020-07-23:
Hi,

The _Monitor_ tab in Cura is used to monitor printers that are directly connected to the computer via a usb cable. However, this functionality is not implemented yet for our Epsilon printers, although is in our roadmap for future version updates. You can use it with the Sigma/Sigmax models. But, for monitoring our Epsilon printers and interacting with them, we have our [cloud](https://cloud.bcn3d.com) platform available.

Regards,
Pedro