BCN3D/Cura #157: 3.2.0 (and earlier) - Right Extruder Only - GCode is wrong for start

WanaGo, 2020-07-02:
**Application Version**
3.2.0

**Steps to Reproduce**
When printing a part with right extruder only, Cura generates GCODE which is incorrect for the start of the print, which causes the right extruder filament to not come out immediately. If you do not use a Brim and cycle around the part a few times, the starting of the actual part will not be printing any filament.
It seems to do a retract but does not feed it back in on the right extruder, meaning there is no filament there when the print starts.

**Actual Results**
Right does not start correctly. Left starts correctly. Look at the generated GCODE you will see the lines which are incorrect, this needs to be fixed.

**Additional Information**
I came from 2.1.5 I think it was (I just formatted my PC) and that works correctly, however 3.2.0 does not work correctly.

If you have the G1 E-6 without G92 E0 below it, it is okay. 
The G92 E0 zeros the extruder when it is retracted -6 so that is why it won't print right away. 
Without zeroing the extruded length it will simply feed that 6 mm back when it starts printing so the real problem is the extra G92 E0

[image]

I believe this has been an issue for some time.

pedro-iatzky, 2020-07-20:
Hi,

I will forward this issue to our team and reach out to you as soon as I have some update.

Thank you,
Pedro

eskeyaar, 2020-09-28:
what if we remove this line from start gcode of right extruder?

eskeyaar, 2020-10-20:
> 
> 
> Hi,
> we are still having problems with printing with right extruder, please fix it ASAP.
No matter, if we print right extruder as first layer in dual printing or it comes after left extrusion printing, .There is always E0.
Whereas, the left extruder has E 6.5 when starting to print after right extruder lays first half of print. Something wrong in the right extruder coding itself.
 
[image]

WanaGo, 2020-10-20:
This was reported in July, please can we get this fixed???? @pedro-iatzky