BCN3D/Cura #159: SigmaR19 with Cura 4.5

eskeyaar, 2020-07-24:
Hi Everyone,
Am planning to use Latest version of Cura ( maybe 4.5), and already sorted out major hurdles, and hopefully, i can start testing with my machine as soon it arrives. Before my machines arrive, if someone can test it out, that would be great.
The reason why I want to use Cura? 
*It has more features and better support generation.
*Also, I think it also solves some of USB stuttering issues due to the feature of max deviation and max resolution which BCN Cura version doesn't have. ( it has but not quite good as latest one)
*And many more fixes

Also , planning to make Prusa slicer or Slic3r version of same. 

Thanks

martinjacala, 2020-09-13:
Were you able to make Cura 4.x work with SigmaR19?

eskeyaar, 2020-09-13:
I'm getting machine tomorrow. So basically it should work.
I have all 2 print modes setuped. 
If you are interested then I can send you some models to test and eventually can share settings and files for cura 4.x

martinjacala, 2020-09-13:
I'd appreciate if you can share the profile/settings for Cura 4.x

Xing-Yun-dev, 2020-09-14:
Just for information: I have triedd with cura 4.6, the .def.json definition file doesnot work instantly, because the combo selector of duplication/mirror/standard is not avaliable. if you want to use stock cura. you can use the standard build volume and involke __M605__  for mode selection.

eskeyaar, 2020-09-14:
@Xing-Yun-dev , @martinjacala 
i have done it from my side ,
[image]
[image]
So dont need to invoke and Mcodes on start gcode.
Will be sharing settings soon,

Thanks

eskeyaar, 2020-09-21:
Sorry, guys, 
having problem in slicing as its conflicts some functions with new cura. 
Will give an update to this later.

S3D-r, 2020-12-04:
> Sorry, guys,
> having problem in slicing as its conflicts some functions with new cura.
> Will give an update to this later.

Any news...?

eskeyaar, 2020-12-05:
Yes, mostly done. Will attach settings so that you guys can test it.

eskeyaar, 2020-12-17:
@S3D-r Will be attaching zip file of all definitions today(hopefully), can you give a shot and let me know how it goes? 
Make sure you have backup of Cura profiles and settings of Cura 4.5. Because it can affect, so just giving info before it goes anything wrong.

Thanks

S3D-r, 2020-12-17:
@eskeyaar  I'd check this out, but I'm pretty busy currently so that It will take some time.

eskeyaar, 2020-12-17:
@S3D-r No problem, Take ur time to test. Maybe other guys can also test it.

eskeyaar, 2020-12-22:
@S3D-r @Xing-Yun-dev @martinjacala @RaulLazaro 

https://github.com/eskeyaar/SigmaR19---Cura-4.6.2
Made repository branch for Cura 4.6.2. Give it a shot. Pls make sure you backup all things before trying. All profiles, All definitions etc..  
Added a base Dual profile, Not added any profile for mirror or Duplication. You can make exact same settings from BCN3D Cura. 
Thanks.