BCN3D/Cura #162: Initial Fan speed height parameter addition

Alphakiew, 2020-11-16:
Hello everyone,
I am not a code guy and I don't really know how things works here so first of all I apologize if I am not at the right place or if I make something wrong on this issue list. Please tell me and I will correct it.
I have a Sigma R17 since 2 years and I have a problem with the fan cooling in Cura : it is not possible to adjust the initial fan speed height that would permit to keep the initial fan speed during "x" layers (or "y"mm).
I heard that this parameter exist on other slicers and it permits to keep no ventilation for many layers to avoid wraping.
Thanks a lot! 
Good continuation !

eskeyaar, 2020-11-16:
Hi,
You can play with these settings here. 
[image]

Alphakiew, 2020-11-16:
Hi Eskeyaar,
Thanks a lot for your answer but this does not answer my problematic. 
Lets do an example : 
I would like the fan to be at 0% from layers 0 to 5 to ensure a good adhesion and then raising fan speed from layer 5 to layer 25 to finally reach the final fan speed at 80%. 
With the parameter regular fan speed at Height/Layer, the fan starts to turn immediately at the 2nd layer and even at 3% the cooling is too high for some materials.
Thanks a lot!

eskeyaar, 2020-11-16:
if you change regular fan speed to 0%, then set fan speed at layer 5, and max speed to 80% rest it will be.

Alphakiew, 2020-11-16:
If I do that I will have a incrementation between 0 and 80% from the first layer 👍 

1st layer : 0%
2nd layer : 20% (here is the problem)
3rd layer : 40%
4th layer : 60%
5th layer : 80%
so the second layer will be with fan > 0% 

I would like the ability to set initial fan speed until layer (layer 5 for the example here with 80% at layer 9) :
1st layer : 0%
2nd layer : 0%
3rd layer : 0%
4th layer : 0%
5th layer : 0%
6th layer : 20%
7th layer : 40%
8th layer : 60%
9th layer : 80%

The idea would be to set the initial fan speed and the layer it stays
and the final fan speed and the layer it should be reached.

eskeyaar, 2020-11-24:
> If I do that I will have a incrementation between 0 and 80% from the first layer
>
> 1st layer : 0%
> 2nd layer : 20% (here is the problem)
> 3rd layer : 40%
> 4th layer : 60%
> 5th layer : 80%
> so the second [...]

[AI3M_soapdish.zip](https://github.com/BCN3D/Cura/files/5589501/AI3M_soapdish.zip)
Something like this you wanted? Note this is just some random machine.

Alphakiew, 2020-11-24:
I looked at it with notepad (I don't have gcode analyser ha ha) but it seems that fans is at 0 (M107) until layer 6 where it turns on (M106 S204).
Yes it seems to fit but it changes from 0 to 204 in one shot from layer 5 to 6 right? 

But don't you understand my need?
I think having these parameters would be perfect : 
- Initial fan speed (already exist)
- Initial fan speed until height (NEW !!! XD)
- Regular fan speed (already exist)
- Regular fan speed at height (already exist)
It would permits to choose the "starting" fan speed and how long it's kept and the final fan speed with the duration of the transition. 
About the "values verification" : Initial fan speed until height < Regular fan speed at height

eskeyaar, 2020-11-24:
Attached Some rough gcode, which is what I think you want to achieve it. 
I belive this version of BCN3DCura doesn't have this feature. If you want to have something like what I did, install this Version of Cura 
https://github.com/smartavionics/Cura/releases
He has built many features which master Cura yet to build it. Feel free to use. 
But if you want to have Sigma machine under this cura, I'm working on files, so you can install my settings under this build to get Sigma machine. 
Currently working on Sigma R19. Soon will add Sigma R17 machine too.

Thanks
[UM2_soapdish.zip](https://github.com/BCN3D/Cura/files/5592107/UM2_soapdish.zip)

Alphakiew, 2020-12-14:
Yes this version doesn't have this feature. 
Is there a way to add it to the next version?

About Cura master, it is interesting but, for now, as Sigma R17 is not ready yet and don't think I can help on this, I will wait.

Thanks a lot for you interest about my issue !
Best regards !

PS : I have a new issue I will raise about the "print infill every".

closed 2020-12-14 by Alphakiew