BCN3D/Cura #164: Is Material Flow limited ?

Alphakiew, 2020-12-16:
Hello everyones !

It seems that the material flow is limited to a value because I tried multiple values above 150% (200%; 250%) and there is no difference in the quantity of material that comes out the nozzle.
The problem is that even if the value turns amber above 150, I though CURA could manage it. Well it's not.
Maybe it could turn red at the maximum value cura is able to use?

To solve the issue I had to cheat on the filament diameter.

Thanks a lot!

marcmateo, 2020-12-22:
Hello,

Indeed, the Cura internally limits the maximum material flow to 150, and a minimum of 5, and from this value it turns orange even if this number is exceeded. We will study your suggestion for future releases.

Thanks for reporting the issue.

Alphakiew, 2021-01-08:
Ok! Thanks a lot @marcmateo ! 
By the way can you tell me if there is a spot to report issues for futur releases? Did you do it for this one should I do it?
Thanks a lot !
Can I close the issue then?

marcmateo, 2021-01-08:
Hello @Alphakiew ,

At the moment we do not have a specific channel for future suggestions or issues for futur releases, but we will be happy to hear them if you send it to us at support@bcn3d.com.

Kind regards !

closed 2021-01-08 by marcmateo

Alphakiew, 2021-01-08:
It's a pity, it would be easier for you !
Ok, I will do the mail when all my issues will be closed. 
Thanks a lot !